arachnidlabs/reload-pro #80: Positive input is connected to the case of instrument

raivisr, 2015-01-08:
It should be stressed in manual that positive input is shorted to the instrument case via tab on BTS141 and I would suggest that instructions are provided on how to install insulator between power transistor and heatsink for people who are willing to sacrifice power rating for isolation.

limpkin, 2015-01-08:
A few side notes to make users understand how **serious** this issue is.
Users having their power supplies with the DC GND connected to the mains earth may be electrically shocked (60V is starting to be very serious). If bare feet, the user may loop the load high side voltage (up to 60V) to GND through their home's earth.

Arachnid, 2015-01-08:
You're right that this needs to be corrected, and emphasised in the manual. I'll get on it.

60V is still under the low voltage directive lower limit of 75V; in theory it's not dangerous to users, though of course any voltage can be dangerous in the right conditions.

limpkin, 2015-01-08:
Answers from http://electronics.stackexchange.com/questions/19103/how-much-voltage-is-dangerous indicate otherwise.

Arachnid, 2015-01-08:
I really hope that nobody's operating the R:LP or any other electrical equipment while standing in a bath of conductive liquid. However, point taken that it's a potential hazard if not notified.

Here's my current plan:
 - Post an advisory on Kickstarter, by email to purchasers, and on my site with a description of the issue, its hazards, and how to remediate it.
 - Offer insulation kits and instructions to remedy the issue for free (postage incl) to anyone who's previously purchased the R:LP
 - Have insulation kits added to unsold stock of the R:LP

Does this sound appropriate?

raivisr, 2015-01-08:
I suggest that you measure impact of couple of most popular insulators (mica and silicone rubber?) on power dissipation capability of the RL:P, and advise users of what they gain and what they lose if they do the mod.

electrokean, 2015-01-08:
This sounds like a good strategy to me Nick.  Anyone working with voltages over 30V should be taking extra care anyway, but I agree most people would not think that the case would be connected to the positive terminal.

Another factor to consider is the anodising/paintwork provides a level of insulation.  In fact I cannot measure a short via the outside of the case on my production unit, except where the silver screw is visible between the heatsink fins (not even in the tapped holes on the heatsink meant for mounting the fan kit).  On my early "first in line" unit, I can measure it because of a scratch, and the bare metal bracket of my prototype fan kit.  Mind you the breakdown voltage of that "insulation" could be less than 60V, and all it takes is a scratch in the suface... so installing an insulator is the correct action for sure.

It is important that people understand the correct way to insulate the TO220 - adding the insulator isn't effective unless a bush is correctly installed on the screw.  I've seen people make that mistake and blow FET's a few times before realising (no, it wasn't me).

I also note that the BTS141 datasheet doesn't actually mention the tab connection to pin 2.  Grrrr!

raivisr, 2016-02-08:
I think this has been resolved a while ago and can be closed as such.

closed 2016-02-08 by raivisr