arachnidlabs/reload-pro #84: Set current <1A off after calibration

danielbriggs, 2015-02-08:
Hi there,
I've installed V1.9 on my Reload:Pro and performed the calibration several times as per the instruction manual with a Rigol DP832 power supply; the supply was setup correctly and did not go into current limiting mode at any point during calibration or operation.

The calibration is off by quite a considerable amount on <1A settings, and no current flows until around 240mA set current. For me, the 0-1A range is most useful so I'm keen to get this working if anyone has any pointers?
Calibration 1-3A range is really good, with no issues.

I graphed the results:
http://danielbriggs.co.uk/images/ReloadPro_Calibration.PNG

I did some searching around and found mention of a possible fix in V1.9.1, but couldn't find anywhere to download this, V1.9 seemed the latest firmware available to download (some 4 months ago).

Any pointers?

electrokean, 2015-02-08:
Try the manual offset calibration process as noted here:
https://github.com/arachnidlabs/reload-pro/issues/74
and here:
https://github.com/arachnidlabs/reload-pro/issues/28#issuecomment-58252206
Should make a big difference to those sub 1A set points.

danielbriggs, 2015-03-02:
Hi there,
Thanks for that, I finally got around to testing it out again.

The CAL O was set to 12.
The response improved gradually the more I dropped this value.
It's now set to "0" and set point matches actual current (+/- 5mA) down to 180mA. 0-180mA the calibration is still way off.
There is also +/- 20mA of oscillation at a lot of the set points up to 3A which I've tested.

Any ideas to improve stability + current response <180mA?
The range 0-500mA is the main range I would like to be using.

Best Regards,
Dan

electrokean, 2015-03-04:
Hmmm... on both my units adjusting the Cal O value gave me quite good calibration down to about 20mA.  Below 50mA and above 50W I have other equipment that I'd use.
I haven't noticed oscillation on my units, but I also don't tend to run them at low currents or monitor them closely for any length of time.  Nick has mentioned in the issue 85 thread that oscillation at low currents may be due to the background software calibration loop.

danielbriggs, 2015-03-30:
I've just updated the unit to V1.10 and re-ran the calibration (with no reset) and it's very much improved over the full range. I noticed an extra calibration step which might have been added since last time.
After calibration, small errors of around +5mA are observed up to 0.5A, and -5mA on higher ranges 1-3A. This is within spec, so I'm now very happy. Thanks!

closed 2015-03-30 by danielbriggs