arachnidlabs/reload-pro #85: Pulsing current draw

tzanetos, 2015-02-11:
I just set up my reload pro from the original kickstarter today. I downloaded the latest firmware and installed the fankit. First time powering on the screen flashed and then went blank but after re-seating the ribbon cable that part worked fine.

The main problem that im seeing is a continuous pulsing of current of the reload pro instead of a steady current draw. My setup is as follows:
-Reload Pro running Reload Pro v1.9.cyacd
-9V battery connected to the red and black leads. 8.17V with no load 
-20mA selected as the current setpoint.
-Agilent MSO3000 series oscope connected to the battery leads in parallel with the reload pro.

The reload pro current current and voltage are jumping all over the place instead of converging to a single value.

Below is my oscilloscope capture of the phenomenon. Note the time scale and voltage range.

[image]

Arachnid, 2015-02-11:
This is almost certainly an issue with the software loop designed to match actual to measured current. Do you have access to a copy of Windows where you can install PSoC creator? If not, I can provide a custom build of the firmware to test this out.

tzanetos, 2015-02-11:
No not readily available.

What is the blue port for?

Arachnid, 2015-02-11:
The blue port is an accessory port, currently unused.

I'll put together a build that has this feature disabled (or at least toggleable) this evening. Can you confirm that you get less variation (proportionally) at higher currents?

tzanetos, 2015-02-11:
Going pat ~180mA appears to create a dead short, so I only used small
currents at first.

Part of my problem is that my DC power supplies are out of commisiom, but I
can use alkaline and lipo (3s 2600mAh) batteries for now

Arachnid, 2015-02-14:
Sorry for the delay; please try flashing the CYACD file here: https://drive.google.com/file/d/0B_iKMAYU3t1LNWh1RFdWRHVoYnM/view?usp=sharing

You may see offset error between the set and actual current with this test firmware, but it should allow us to check that the oscillations are eliminated.

tzanetos, 2015-03-06:
Yes this seemed to fix the oscillations. I see more of an offset  at lower
currents (<500mA). What is the long term fix/solution for this problem, and
why was mine defective. Have others experienced the same?

Also i registered to receive the thermal insulation kit. When will that be
sent out?

Thanks,
-Teddy

brearto2, 2015-09-22:
@Arachnid 
Is there a permanent fix for this issue?

I just downgraded from firmware version 1.11 to fix this issue (using the firmware linked above) because I was getting the same issue. My multimeter was reading upwards of 4A (no oscope available) before my power supply would shutoff.